dotchen/WorldOnRails #2: Some import erros.

SunHaoOne, 2021-05-26:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!

Hello,
    Thanks for you code here. I have downloaded the files and successfully evaluated this model. The final results shows excellent performance. However, I met some problems when training the model  from the beginning.  
### 
**1. Import errors**

There are some import errors, such as `  from common.augmenter import augment
ModuleNotFoundError: No module named 'common'.` 
For example, I have fixed some errors by adding `sys` to find the import files:
```
import ray
import math
import numpy as np
import yaml
import torch
import torch.nn.functional as F
from torch import nn, optim
from models import EgoModel, CameraModel
from bellman import BellmanUpdater
from datasets.main_dataset import MainDataset
import sys
sys.path.append("/home/shy/WorldOnRails-nomodel/utils")
import __init__ 
```
It would be more convenient if you could fix these problems. 
**2. How to use your released dataset**

After downloading your dataset, and I tried to trained the models. However, it fails to load these data, and I have checked the FAQ.md, which shows that we need to write  custom data loader. I'm sorry I'm not familiar with carla data loading. Can you provide some ideas on how to use your release dataset？ I would appreciate it if you could provide this code here.

**3. What does the act_val mean?**
In `rails.py` line 55, I am confused with this act_val. Why it has dimension 4, my understanding is that this action is the three dimensions of steering, throttle and brake and discretize them and finally by softmax to see the maximum probability for the action at this point.

dotchen, 2021-05-26:
> 1. Import errors

What you posted is not the right way to do it. If you execute `python -m xxx` at the root directory of this repo, you do not need to add those additional imports. **Please do follow the instructions provided in RAILS.md.**

> 2. How to use your released dataset

As described in FAQ, you need to write your own data loader. Or you can collect your own dataset using the scripts provided. The instructions of how to do it are also provided in the RAILS.md.

> 3. What does the act_val mean?

`num_speeds=4` in config.yaml

closed 2021-05-26 by dotchen

SunHaoOne, 2021-05-26:
>     1. n?

Thank you for your  quick reply. I come to realize that  I forgot to run the code in the root files, so some import can be failed to load. Because of the import error, I failed to generate the data from Carla, so I turned to download your release datasets. Thanks you again for your reply, I will read your markdown more carefully.

SunHaoOne, 2021-05-26:
> > 1. Import errors
> 
> What you posted is not the right way to do it. If you execute `python -m xxx` at the root directory of this repo, you do not need to add those additional imports. **Please do follow the instructions provided in RAILS.md.**
> 
> > 1. How to use your released dataset
> 
> As described in FAQ, you need to write your own data loader. Or you can collect your own dataset using the scripts provided. The instructions of how to do it are also provided in the RAILS.md.
> 
> > 1. What does the act_val mean?
> 
> `num_speeds=4` in config.yaml

Thank you for your quick reply. I come to realize that I forgot to run the code in the root files, so some import can be failed to load. Because of the import error, I failed to generate the data from Carla, so I turned to download your release datasets. Thanks you again for your reply, I will read your markdown more carefully.