dotchen/WorldOnRails #3: Small error with 他和

SunHaoOne, 2021-05-27:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!

dotchen, 2021-05-27:
> 1 datasets lose key and value

I double-checked and it is a bug in my conversion script. **Thank you for letting me know and I will fix it and update the dataset soon**. In the meantime you can use the data collection script to generate data.

> 2 what does 6+:lane centers mean? 

Those are the lanes for lane changing

> 3 Is Vulcan necessary for Carla?

No, but it runs much faster in Carla than OpenGL. The errors you see however is because there is no results folder. Simply `mkdir results` should do the trick.

Closing the issue for now.

closed 2021-05-27 by dotchen

dotchen, 2021-05-28:
@SunHaoOne I have updated the dataset link, should be fixed. Let me know if you have any more questions.

SunHaoOne, 2021-05-28:
> I double-checked and it is a bug in my conversion script. **Thank you for letting me know and I will fix it and update the dataset soon**. In the meantime you can use the data collection script to [...]

Thanks for your reply. Besides, I found there are 200 lbl images, however, the number of rgb cameras is only 195. For example, the last images of wide camera is`wide_2_00194.jpg`, and the last images of wide segment is `wide_sem_2_00199`

dotchen, 2021-05-28:
`num_plan=5`

varunjammula, 2021-06-01:
@SunHaoOne Hey, were you able to make the data collector work? If you enable the wandb, you can see the logs for rails.data_phase0. I did that and I can see two errors coming up.

[image]
[image]

SunHaoOne, 2021-06-05:
> @SunHaoOne Hey, were you able to make the data collector work? If you enable the wandb, you can see the logs for rails.data_phase0. I did that and I can see two errors coming up.
> 
> [image]
> [image]
> 
> There is no speed sensor in CARLA_0.9.10.1. Should I make any API changes to use the Speed sensor?
> 
> @dotchen Any suggestions here?

Sorry for the late reply. I also meet this problem now, do you have any idea？（I successfully collected data for about 1400 frames before, now I cannot collect data, and I have tried to run carla again but also failed ）It shows as follows:
```
    control = self.run_step(input_data, timestamp)
  File "autoagents/collector_agents/random_collector.py", line 124, in run_step
    self.flush_data()
  File "autoagents/collector_agents/random_collector.py", line 67, in flush_data
    'vid': wandb.Video(np.stack(self.rgbs).transpose((0,3,1,2)), fps=20, format='mp4')
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/wandb/sdk/data_types.py", line 1036, in __init__
    self.encode()
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/wandb/sdk/data_types.py", line 1074, in encode
    clip.write_videofile(filename, **kwargs)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/moviepy/decorators.py", line 54, in requires_duration
    return f(clip, *a, **k)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/moviepy/decorators.py", line 135, in use_clip_fps_by_default
    return f(clip, *new_a, **new_kw)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/moviepy/decorators.py", line 22, in convert_masks_to_RGB
    return f(clip, *a, **k)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/moviepy/video/VideoClip.py", line 307, in write_videofile
    logger=logger)
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/moviepy/video/io/ffmpeg_writer.py", line 216, in ffmpeg_write_video
    ffmpeg_params=ffmpeg_params) as writer:
  File "/home/shy/anaconda3/envs/world_on_rails/lib/python3.7/site-packages/moviepy/video/io/ffmpeg_writer.py", line 88, in __init__
    '-r', '%.02f' % fps,
TypeError: must be real number, not NoneType
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/shy/WorldOnRails0603/leaderboard/leaderboard/leaderboard_evaluator.py", line 365, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/home/shy/WorldOnRails0603/leaderboard/leaderboard/scenarios/scenario_manager.py", line 136, in run_scenario
    self._tick_scenario(timestamp)
  File "/home/shy/WorldOnRails0603/leaderboard/leaderboard/scenarios/scenario_manager.py", line 159, in _tick_scenario
    raise AgentError(e)
leaderboard.autoagents.agent_wrapper.AgentError: must be real number, not NoneType 
```

varunjammula, 2021-06-05:
It looks like you are passing the fps argument working! Check the value properly! Also try to pull the latest code from the repo.

SunHaoOne, 2021-06-05:
> It looks like you are passing the fps argument wrong! Check the value properly! Also try to pull the latest code from the repo.

Thanks for your reply, I run carla with `./carla.sh 2 2 2000` and also with the runners = 2, I can use it for collecting the data in the phase 12, ( I have collected about 140GB for 3 days). After pull the latest repo, I still failed and I will check the code again.
```
#!/bin/bash

num_servers=$1
num_gpus=$2
base=$3
for (( i=1; i<=$1; i++ ))
do
    port=$((i*base))
    fuser $port/tcp -k
    fuser $((port + 1))/tcp -k
    fuser $((port + 2))/tcp -k
	DISPLAY= CUDA_VISIBLE_DEVICES=$(((i-1)%num_gpus)) /home/shy/Desktop/carla0.9.10.1/CarlaUE4.sh -world-port=$port -opengl -trafficmanager-port=$((port+2)) &
done
wait
```

dotchen, 2021-06-05:
@SunHaoOne reinstall `moviepy`, `imageio` and `wandb`

SunHaoOne, 2021-06-06:
> @SunHaoOne reinstall `moviepy`, `imageio` and `wandb`

Thanks you again! Wow, I fixed it and now I can collected the data.