dotchen/WorldOnRails #5: Small issues in running code

aaronh65, 2021-05-28:
Hey Dian,

I'm working with your code right now and I thought I'd quickly point out some small issues I encountered while working on it. These are mostly small things and I may be missing other small issues I found:

1.  during environment setup, I had to do the following to get the right environment: (1) create conda environment from environment.yaml file (as stated in INSTALL.md), (2) `pip install -r requirements.txt` for the requirements files in `scenario_runner` and `leaderboard` directories, (3) uninstall the CARLA 0.9.5 that gets installed as part of one of the `requirements.txt` files (`pip uninstall carla`), (4) `easy_install` the CARLA .egg file that comes with the CARLA 0.9.10.1 repo
2. the carla server launch script doesn't exactly line up with the INSTALL.md specification (accepts $2 and $3 args but there's no $1 arg?). I rewrote the script to add a third argument that specifies how many GPUs you're running with, and sets CUDA_VISIBLE_DEVICES appropriately. So for 4 runners and 4 GPUS, it starts 1 CARLA server per GPU
3. `python -m rails.data_phase2` isn't able to import `WORActionLabeler` and running a `grep` from the repo root doesn't turn up any code that has that. I did find a `RAILSActionLabeler` though, which I'm assuming is what is supposed to be imported?

I'll update this post/thread if I find anything else!

dotchen, 2021-05-28:
Thanks for pointing these out!

1. I think the `environment.yaml` should contain all the requirements and you should not need to do additional pip installs. Let me know otherwise however and I will add the missing dependencies in the `environment.yaml` file.
2. fixed, that was a refactoring typo
4. good catch! the original method name is 'WOR' and I forgot to change that one :P

dotchen, 2021-05-29:
As of the latest commit, everything should be fixed. Feel free to create a new issue if you find out other typos.

closed 2021-05-29 by dotchen