dotchen/WorldOnRails #8: Error in data collection - data_phase0.py

HimangiM, 2021-06-01:
Hi,

When I try to run the data_phase0.py, I am getting an error - TypeError: missing a required argument: 'route' at the line 16, ScenarioRunner.remote function. One of the arguments in the remote function seems missing in the remote function. Maybe, it is scenario_class. Can you please let me know how to add the scenario_class argument in the function? Is it similar to that of data_phase1.py?

Any help would be greatly appreciated.

Thanks,
Himangi

aaronh65, 2021-06-01:
Hey Himangi,

I forgot to include this in my other "bug list" issues, but I resolved this by hardcoding a `scenario_class = train_scenario` variable and passing that in as the second argument to `ScenarioRunner.remote`. @dotchen does that seem reasonable? I suppose this could also be an argparse variable (possibly if you wish to run on NoCrash scenarios). For the ego model/`data_phase0`, I don't think NoCrash vs Leaderboard scenarios matter all that much, the data agent is still moving randomly.

dotchen, 2021-06-01:
@aaronh65  yes, except that the scenario class should be `route_scenario`, so it would use the Town06 highway as the environment. I have fixed that in `data_phase0.py`, sorry for the inconvenience.

closed 2021-06-01 by dotchen