dotchen/WorldOnRails #9: Speed sensor and error while collecting data

varunjammula, 2021-06-02:
@SunHaoOne Hey, were you able to make the data collector work? If you enable the wandb, you can see the logs for rails.data_phase0. I did that and I can see two errors coming up.

[image]
[image]

There is no speed sensor in CARLA_0.9.10.1. Should I make any API changes to use the Speed sensor?

@dotchen Any suggestions here?

_Originally posted by @varunjammula in https://github.com/dotchen/WorldOnRails/issues/3#issuecomment-851847004_

dotchen, 2021-06-02:
Please use the latest commit which has fixed the typo in `random_collector`.

closed 2021-06-02 by dotchen

varunjammula, 2021-06-02:
It works, thanks!