dotchen/WorldOnRails #12: VisualizationDataset Class in main_dataset.py

SunHaoOne, 2021-06-04:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!
Hey Dian,
  I successfully collected the dataphase1 by installing the torch 1.4.0, and I am confused by these function in the main_dataset.py.
### 1. Maindataset Class
```
full_path='/home/shy/Desktop/WorldOnRails/DataAndModel/PhaseData_12/'
config_path = '/home/shy/Desktop/WorldOnRails/config.yaml'
dataset = MainDataset(full_path, config_path)
wide_rgb, wide_sem, narr_rgb, lbls, locs, rots, spds, cmd = dataset[30]
```
Then I can load these data, and I see it returns the right value. So do I need to delete the small dataset by myself?
```
/home/shy/Desktop/WorldOnRails/DataAndModel/PhaseData_12/udmprkyurt  is too small. consider deleting it.
/home/shy/Desktop/WorldOnRails/DataAndModel/PhaseData_12/: 17768 frames (x3)
```
Besides, the dataset collected includes wide_rgb, wide_sem and narr_rgb ,but there is no narr_sem here. And I found `narr_sem` is in the Class `LabeledMainDataset`. So is `narr_sem` is necessary for training?
### 2. VisualizationDataset Class and LabeledMainDataset
I failed to run these class, which returns `a bytes-like object is required, not 'NoneType' `, and I checked the class in this respository, it seems the class is not adpoted. I guess these class are used to generate the rgb sem images.
```
Vdataset = VisualizationDataset(full_path, config_path)
idx=30
lmdb_txn = Vdataset.txn_map[idx]
index = Vdataset.idx_map[idx]

locs = Vdataset.__class__.access('loc', lmdb_txn, index, 6, dtype=np.float32)
rots = Vdataset.__class__.access('rot', lmdb_txn, index, 5, dtype=np.float32)
spds = Vdataset.__class__.access('spd', lmdb_txn, index, 5, dtype=np.float32).flatten()
lbls = Vdataset.__class__.access('lbl', lmdb_txn, index+1, 5, dtype=np.uint8).reshape(-1,96,96,12)
#maps = Vdataset.__class__.access('map', lmdb_txn, index+1, 5, dtype=np.uint8).reshape(-1,1536,1536,12)

#rgb = Vdataset.__class__.access('rgb',  lmdb_txn, index, 1, dtype=np.uint8).reshape(720,1280,3)
cmd = Vdataset.__class__.access('cmd', lmdb_txn, index, 1, dtype=np.float32).flatten()
#act = Vdataset.__class__.access('act', lmdb_txn, index, 1, dtype=np.float32).flatten()
```
After testing, I found maps, rgb and act can not be loaded successfully. (Might the class is no needed). Thank you again!

dotchen, 2021-06-04:
I am a bit lost... Can you tell me which step (from the readme) you are trying to do here?

SunHaoOne, 2021-06-04:
> I am a bit lost... Can you tell me which step (from the readme) you are trying to do here?

Thanks for your reply. Now I am running `python -m rails.data_phase1 --scenario=nocrash_train_scenario --num-runners=4 --port=2000`, then I see some data saved, and I tried to see the data in this data.dbm. Then in this py file, I see how your code loaded the data. And other class cannot be used for some reason (maybe it is not used for the following trainning) , for example, this seg images is gray, (the shape of wide_sem is(192, 480)), however in your released data, the shape is 3 channels.
```
print("locs=",locs)
print("rots=",rots)
print("spds=",spds)
print("cmd=",cmd)

locs= [[129.24919   86.07171    8.401869]
 [129.30621   87.032166   8.338286]
 [129.36586   88.04143    8.265622]
 [129.4187    89.02836    8.196826]
 [129.34424   90.01089    8.129512]
 [129.32495   91.06913    8.054202]]
rots= [[86.60031 ]
 [86.605354]
 [86.61064 ]
 [86.83437 ]
 [90.87115 ]]
spds= [3.7831488 3.9974685 4.019558  3.912955  4.169325 ]
cmd= 3
```

dotchen, 2021-06-04:
>  for example, this seg images is gray, (the shape of wide_sem is(192, 480)), however in your released data, the shape is 3 channels.

They are not 3 channel files, they are stored as palete PNGs, as described in the [README](
https://github.com/dotchen/WorldOnRails/blob/release/docs/DATASET.md#semantic-labels).

closed 2021-06-04 by dotchen

SunHaoOne, 2021-06-04:
> They are not 3 channel files, they are stored as palete PNGs, as described in the [README](https://github.com/dotchen/WorldOnRails/blob/release/docs/DATASET.md#semantic-labels).

Thank you very much for your patient reply. And I understand the data by reading readme again. And may I ask you a simple question?
After reading your code in rails.py,  
```
 def train_ego(self, locs, rots, spds, acts):               
 locs = locs[...,:2].to(self.device)       
 yaws = rots[...,2:].to(self.device) * math.pi / 180.       
 spds = spds.to(self.device)        
 acts = acts.to(self.device)                
pred_locs = []   pred_yaws = []             
pred_loc = locs[:,0]       
pred_yaw = yaws[:,0]      
pred_spd = spds[:,0]       
for t in range(self.ego_traj_len-1):         
   act = acts[:,t]                        
   pred_loc, pred_yaw, pred_spd = self.ego_model(pred_loc, pred_yaw, pred_spd, act)
```
I tried to load the data and found the acts is as follows:
```
array([[ 0.4324325 ,  0.5298194 ,  0.        ],
       [ 0.87813395,  0.41267335,  0.        ],
       [-0.251706  ,  0.855824  ,  0.        ],
       [-0.26127693,  0.29226604,  0.        ],
       [ 0.64959806,  0.8846799 ,  0.        ],
       [ 0.        ,  0.        ,  1.        ],
       [ 0.21413343,  0.8341027 ,  0.        ],
       [-0.59543383,  0.08719549,  0.        ],
       [-0.41294557,  0.1233392 ,  0.        ],
       [ 0.8560932 ,  0.05150029,  0.        ]], dtype=float32)

```
And this ego_model means, input is 10 history locs, yaws, speeds and actions and output is locs, yaws and speeds. In this loop, action is a list of [throttle, steer and brake], then in the next loop, the input is two lists of [throttle, steer and brake], act = acts[:,t]   ,and finally  input 5 actions of 10 and output the 10 lists of [throttle, steer and brake]. However, I am not sure whether  `act = acts[:,t] `   needs to change to `act = acts[: t]`. For example, there is no dim for `acts[:,4]` because it only has 3 dim here. And then   `steer = acts[...,0:1],throt = acts[...,1:2],brake = acts[...,2:3]` can return the right value. Thank you again！