dotchen/WorldOnRails #21: Continuously writing data to disk rather than doing it all in one go

aaronh65, 2021-06-29:
I asked about the set_synchronous_mode timeout issue in the CARLA repository and it might be a lot of hassle to address.

I'd like to instead try changing how the data is written to disk during `data_phase1`. Currently, various arrays keep track of RGB images and other data and `lmdb` writes them to disk at the end of the run or after collision. Is it possible to use `lmdb` to continuously write data to disk during the run, rather than doing it all at the end? 

I'm not very familiar with the library so any help is appreciated!

dotchen, 2021-06-30:
The easiest way is to set `num_per_flush` to 1 in `config.yaml`

aaronh65, 2021-06-30:
I ended up taking a slightly different approach and added lmdb_env and txn as class attributes of the collector agent. I setup an init function that runs during the first run_step to initialize the above, and upon collision closes the lmdb_env. While running, the write txn continuously puts data into the database.

Fingers crossed that this fixes the timeout bug I've been seeing

closed 2021-06-30 by aaronh65

dotchen, 2021-07-01:
Cool, let me know if this solves your problem!

aaronh65, 2021-07-01:
Data collection seems to be much more stable so far. I think one reason this happens on my setup is because I'm using a lab cluster and writing over NFS to a centralized HDD storage for large datasets located on a different node, rather than writing to the data collector node's SSD. Perhaps writing Gbs of data at the end of a run to a remote HDD slows the system down and causes timeouts?