dotchen/WorldOnRails #22: lbc_agent

coolCoco-MiMi, 2021-07-02:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!

Hi,

I want to test the lbc_agent and changed the "--agent=autoagents/lbc_agent" for LBC. But when I was creating the "config_lbc.yaml", I could not find the weights for the LBC agent. So, is it provided? Or can you provide the weights for running the lbc agent?

dotchen, 2021-07-02:
Hello,

Thank you for your interest in this project. Please use [lbc's official repo](https://github.com/dotchen/LearningByCheating) if you would like to compare to LBC, which contains pretrained model weights.

You can use this repo's LBC implementation if you want to train LBC on scenario runner, or CARLA 0.9.10.

closed 2021-07-02 by dotchen

coolCoco-MiMi, 2021-07-02:
Thanks for replying. 

I see. But that pertained weights in the lbc's official repo cannot be used in this structure right? I can only train LBC for the scenario runner?

dotchen, 2021-07-05:
That's correct.

You can also train the LBC implementation in this repo for NoCrash. See [here](https://github.com/dotchen/WorldOnRails/blob/release/docs/RAILS.md#data-collection) more further details.