dotchen/WorldOnRails #24: Issues with running the validation routes

aaronh65, 2021-07-08:
Hey Dian,

I've noticed an issue when deploying agents on the validation routes. On some of them, non-ego agents will be stuck in front of the ego agent which causes it to halt until timeout. The following shows route completion/driving score of the driving agent on all validation routes, each run for 5 repetitions

[image]

Some of them, like route_09 and route_20 reach the same score every single time. Taking a look at route_20's debug videos, you see stuff like this

[image]

This vehicle just sits there for ages without doing anything, and causes the ego agent to timeout even though the ego agent didn't do anything wrong. Have you seen anything like this in your own experiments?

dotchen, 2021-07-08:
Thanks for the info! The evaluation numbers are very interesting. Could you explain more on the setup?

Regarding vehicles getting stuck: This could be one of the scenarios (scenario 6 if I remember correctly), where you are expected to change lane and pass the vehicle.

aaronh65, 2021-07-09:
For evaluation, we iterate through the `routes_testing` folder and run each route for 4 repetitions. For each route, we record the mean and standard deviation for route completion and driving score to create the above plot.

I wasn't aware that vehicle passing was explicitly modeled in Leaderboard, I worked a lot with CARLA 0.9.11 and didn't see any of those scenarios then (not sure if it's a version issue?). I'll close this issue then if this is expected behavior

dotchen, 2021-07-09:
Cool. Is this with the provided weights?

aaronh65, 2021-07-11:
Ah I see, it looks like the `all_towns_traffic_scenarios.json` file included in Brady Zhou's [LBC codebase](https://github.com/bradyz/2020_CARLA_challenge) only has scenarios 1, 3, 4... that's the codebase I built off of. I guess that explains why I never saw Scenario 06 before

dotchen, 2021-07-11:
> Ah I see, it looks like the all_towns_traffic_scenarios.json file included in Brady Zhou's LBC codebase only has scenarios 1, 3, 4... that's the codebase I built off of. I guess that explains why I never saw Scenario 06 before working with your code.

That makes sense. Marking as resolved.

closed 2021-07-11 by dotchen