dotchen/WorldOnRails #27: Origin leaderboard submit evaluate

Kin-Zhang, 2021-08-04:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!

When I tried to use pretrained model to origin leaderboard, the `image_agent.py` sensors config is not correct with WOR team submit on leaderboard benchmark
[image]

And for the image_agent posted on this repo:
1. the collision and stitch_camera is not allowed in origin leaderboard
2. the width and height is fixed by origin leaderboard
```python
        sensors = [
            {'type': 'sensor.collision', 'id': 'COLLISION'},
            {'type': 'sensor.speedometer', 'id': 'EGO'},
            {'type': 'sensor.other.gnss', 'x': 0., 'y': 0.0, 'z': self.camera_z, 'id': 'GPS'},
            {'type': 'sensor.stitch_camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
            'width': 160, 'height': 240, 'fov': 60, 'id': f'Wide_RGB'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
            'width': 384, 'height': 240, 'fov': 50, 'id': f'Narrow_RGB'},
        ]
```

I want to see from the origin leaderboard because I want to know about the failure situation on WOR and LBC since the result analysis about fail is lack.

If it's possible could you pls give the config/agent and pretrained model files submit to leaderboard (like using another branch for origin leaderboard) or upload the result files after submiting leaderboard , either one is ok for analysis fail situation.

Kin-Zhang, 2021-08-04:
The **analysis** I mean here is like this:
what's kind of situation is difficult for WOR, what's kind of situation is LBC cannot solve but WOR can solve? (from the detailed result comparison)

Kin-Zhang, 2021-08-04:
And I also try to modify the sensor config to satisfy the origin leaderboard
```python
        sensors = [
            # {'type': 'sensor.collision', 'id': 'COLLISION'},
            {'type': 'sensor.speedometer', 'id': 'EGO'},
            {'type': 'sensor.other.gnss', 'x': 0., 'y': 0.0, 'z': self.camera_z, 'id': 'GPS'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
            'width': 160, 'height': 240, 'fov': 60, 'id': f'Wide_RGB'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
            'width': 384, 'height': 240, 'fov': 50, 'id': f'Narrow_RGB'},
        ]
```
And change the speed sensor
```python
`spd = ego['speed']`
```
but the result is that car don't move or straight to the other road like: I believe is might be the different setting you evaluate to origin leaderboard

https://user-images.githubusercontent.com/35365764/128114109-a9c7f830-4f33-4beb-b08d-fd02b652104a.mp4

https://user-images.githubusercontent.com/35365764/128113935-1d986231-c581-4cae-90ed-f6a51ee9bb0d.mp4

The evaluation is just from carla leaderboard get start: https://leaderboard.carla.org/get_started/
`./scripts/run_evaluation.sh`

dotchen, 2021-08-04:
> the collision and stitch_camera is not allowed in origin leaderboard

If you want to use the pretrained weight to submit to the leaderboard, you need to modify the `image_agent.py`, as `image_agent.py` in this repo is built for the `leaderboard` provided.

> the width and height is fixed by origin leaderboard

You can specify height, width, camera fov, or even the camera xyz location. They are not fixed.

> upload the result files after submiting leaderboard , either one is ok for analysis fail situation.

Unfortunately I don't think the online leaderboard provide raw statistics for each run as that would provide side-channel info for one to cheat. You might want to evaluate on the training or validation routes for proxy.

> And I also try to modify the sensor config to satisfy the origin leaderboard

Judging from the video I think you specified the camera parameters wrong (P.S Also don't use any debug option since that black arrow is not supposed to be in any images.). Check the demo video on our [website](https://dotchen.github.io/world_on_rails/) for reference.

closed 2021-08-04 by Kin-Zhang

Kin-Zhang, 2021-08-04:
Thanks for reply,
> Judging from the video I think you specified the camera parameters wrong (P.S Also don't use any debug option since that black arrow is not supposed to be in any images.). Check the demo video on our website for reference.

on the website video it seems like you just use one camera, but not like leaderboard submit which is four?

> If you want to use the pretrained weight to submit to the leaderboard, you need to modify the image_agent.py

So the way I just modified sensor config is not complete correct way?

dotchen, 2021-08-04:
> on the website video it seems like you just use one camera, but not like leaderboard submit which is four?

On the website video there is no black arrows on contrary to the ones you showed.

> So the way I just modified sensor config is not complete correct way?

You should just replace the `stitch camera` with the equivalent three cameras.

dotchen, 2021-08-04:
Regarding on how the cameras are configured: please also check the description in section 5 of [our paper](https://arxiv.org/pdf/2105.00636.pdf), if you are having trouble finding it in the repo.

Kin-Zhang, 2021-08-04:
> Regarding how the cameras are configured: please also check the description in section 5 of [our paper](https://arxiv.org/pdf/2105.00636.pdf), if you are having trouble finding it in the repo.

Thanks Again 💖💖💖

varunjammula, 2021-08-23:
> And I also try to modify the sensor config to satisfy the origin leaderboard
> 
> ```python
>         sensors = [
>             # {'type': 'sensor.collision', 'id': 'COLLISION'},
>             {'type': 'sensor.speedometer', 'id': 'EGO'},
>             {'type': 'sensor.other.gnss', 'x': 0., 'y': 0.0, 'z': self.camera_z, 'id': 'GPS'},
>             {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
>             'width': 160, 'height': 240, 'fov': 60, 'id': f'Wide_RGB'},
>             {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
>             'width': 384, 'height': 240, 'fov': 50, 'id': f'Narrow_RGB'},
>         ]
> ```
> 
> And change the speed sensor
> 
> ```python
> `spd = ego['speed']`
> ```
> 
> but the result is that car don't move or straight to the other road like: I believe is might be the different setting you evaluate to origin leaderboard
> 
>  vid_12_05f137fe6c069f9fecbb.mp4 
>  vid_4_fc1e5db93e0c0a5b952f.mp4 
> The evaluation is just from carla leaderboard get start: https://leaderboard.carla.org/get_started/
> `./scripts/run_evaluation.sh`

Hey, I think one issue here might be that you are running the evaluation with debug=True. If that is the case, the image input is corrupted by the waypoint markers and the model does not work.

Kin-Zhang, 2022-03-08:
> Hey, I think one issue here might be that you are running the evaluation with debug=True. If that is the case, the image input is corrupted by the waypoint markers and the model does not work.

Yes, After I detailed evaluated my model, I checked this issue again, I knew the problem now. 

But it seems the pre-trained model provided on repo readme is not great. I success evaluated it on just one route in devtest.xml about Town01, it seems weird maybe, it needs to train... more

Here is a detailed eval video for people who want to know more:

https://user-images.githubusercontent.com/35365764/157237077-67e88dcc-3172-49ac-a1f5-6945afb319a8.mp4

I thought the problem may be the ego model .... maybe, and here is the eval image in docker hub.

```bash
docker pull zhangkin/wor
docker run -it --net=host --gpus all zhangkin/wor /bin/bash
```
and then in container run this after open CARLA in port 2000
```bash
./leaderboard/scripts/run_evaluation.sh
```
select login `wandb` or don't visualize results by inputting `3`.

BTW: The modal file in docker images is all provided from this repo pre-trained.

dotchen, 2022-03-08:
Is the camera setup the same as what is configured in the repo by default? Judging from the video above it does not seem to match my original setup.

Kin-Zhang, 2022-03-09:
> Is the camera setup the same as what is configured in the repo by default? Judging from the video above it does not seem to match my original setup.

Oh! thanks for replying! I notice that you will release new paper Learning from all vehicle, really looking forward to that.

and for setup, only `stitch_camera` to `camera`, other config about setting is all from ==yaml config file== in model file provided.

Kin-Zhang, 2022-03-10:
Thanks for replying! Finally, I knew the problem from, the config yaml file didn't point out the camera number which you copied here is four. And I just followed the `image_agent.py` file which point the camera number is 2, one is Wide and another one is narrow.

> Make sure it matches the configuration below:

After you point out the sensor config here, and with modified at [`image_agent.py` ](https://github.com/dotchen/WorldOnRails/blob/release/autoagents/image_agent.py#L87)with wide_rgb , following codes:
```python
for i in range(3):
    _, wide_rgb = input_data.get(f'Wide_RGB_{i}')
    _wide_rgb = wide_rgb[...,::-1].copy()[...,:3]
    wide_rgbs.append(_wide_rgb)
wide_rgbs_con = np.concatenate([wide_rgbs[0],wide_rgbs[1],wide_rgbs[2]], axis=1)
wide_rgbs_ = torch.tensor(wide_rgbs_con[None]).float().permute(0,3,1,2).to(self.device)
steer_logits, throt_logits, brake_logits = self.image_model.policy(wide_rgbs_, _narr_rgb, cmd_value)
```
I run the routes again but with ... same effects. Is there any things I did wrong where I modified?

https://user-images.githubusercontent.com/35365764/157585824-b61aaac6-1904-42c7-b76d-c42a09fa61c5.mp4

dotchen, 2022-03-10:
I think you are missing the cropping: https://github.com/dotchen/WorldOnRails/blob/release/autoagents/image_agent.py#L103-L105

Kin-Zhang, 2022-03-10:
> I think you are missing the cropping: https://github.com/dotchen/WorldOnRails/blob/release/autoagents/image_agent.py#L103-L105

Thanks! The reason why I missing it is that the concatenate action to visual wandb will lead the size be error and I forgot that also those crop images to model.

I added it here:
```bash
        for i in range(3):
            _, wide_rgb = input_data.get(f'Wide_RGB_{i}')
            wide_rgb_crop = wide_rgb[self.wide_crop_top:,:,:3]
            _wide_rgb = wide_rgb_crop[...,::-1].copy()
            wide_rgbs.append(_wide_rgb)

        _, narr_rgb = input_data.get(f'Narrow_RGB')
        narr_rgb_crop = narr_rgb[:-self.narr_crop_bottom,:,:3]
        _narr_rgb = narr_rgb_crop[...,::-1].copy()
```
And the cars went enough for evaluating comparison in local mode but still stack when turning too much, I thought maybe it needs to train more to perform better in junction turning. Here is the video and result table for devtest.xml Town01 routes for others who are interested:

[image]

https://user-images.githubusercontent.com/35365764/157628022-37a9e677-fec6-4eb9-9a56-b7b4c32081d0.mp4

https://user-images.githubusercontent.com/35365764/157628035-160fecd6-8c13-4aff-80c6-e5c0be676222.mp4

Thanks again for your help. Really appreciate. If I still missing something on this evaluation, please let me know.

Looking forward to your new paper and codes!

dotchen, 2022-03-10:
This still looks odd. Can you run the model with codes provided in the repo verbatim, and see how it works?

> Looking forward to your new paper and codes!
Really appreciate your interest. We will release the paper and code by the end of March.

Kin-Zhang, 2022-03-11:
> This still looks odd. Can you run the model with codes provided in the repo verbatim, and see how it works?

the code provided in the repo need modify a little since camera number is different like I tried:
1. `spd` to `speed`
2. `stitch_camera ` to `camera`
3. and the number of camera with modify codes above.

I thought maybe the **ego model** provided in the repo is for **two cameras** model, for four cameras it could be odd as shown in the video? I will try to check the `image_agent.py` to see whether I missed something.

> We will release the paper and code of "Learning from All Vehicles (LAV)" by the end of March.

Thanks for letting me know! 🤩

Kin-Zhang, 2022-03-11:
I think I found the reason now. The correct camera `height` should be `240`. After I detailed visited the codes and config here https://github.com/dotchen/WorldOnRails/issues/27#issuecomment-1063092131

here is the sensor config code:
```bash
        sensors = [
            {'type': 'sensor.speedometer', 'id': 'EGO'},
            {'type': 'sensor.other.gnss', 'x': 0., 'y': 0.0, 'z': self.camera_z, 'id': 'GPS'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': -55.0,
            'width': 160, 'height': 240, 'fov': 60, 'id': f'Wide_RGB_0'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
            'width': 160, 'height': 240, 'fov': 60, 'id': f'Wide_RGB_1'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw':  55.0,
            'width': 160, 'height': 240, 'fov': 60, 'id': f'Wide_RGB_2'},
            {'type': 'sensor.camera.rgb', 'x': self.camera_x, 'y': 0, 'z': self.camera_z, 'roll': 0.0, 'pitch': 0.0, 'yaw': 0.0,
            'width': 384, 'height': 240, 'fov': 50, 'id': f'Narrow_RGB'},
        ]
        return sensors
```

Finally, it looks really great now. Thanks! Really appreciate. 🥰🥰

dotchen, 2022-03-11:
Nice carch. Glad it works now.