dotchen/WorldOnRails #30: python -m rails.data_phase0 --num-runners=[NUM RUNNERS] --port=[WORLD PORT]   exit no outdata; when change ray local_mode=True ok

zdx3578, 2021-10-04:
i run carla and worldinrails in server ;
run carla:     DISPLAY= ${CARLA_ROOT}/CarlaUE4.sh  -world-port=$port  -opengl -quality-level=low -world-port=$port -resx=400 -resy=300  $    change  in file  scripts/launch_carla.sh

pygame no screen change settint :
+++ b/leaderboard/leaderboard/leaderboard_evaluator.py
@@ -23,7 +23,7 @@ import pkg_resources
 import sys
 import carla
 import signal
-
+os.environ['SDL_VIDEODRIVER'] = 'dummy'

when run:   python -m rails.data_phase0 --num-runners=[NUM RUNNERS] --port=[WORLD PORT]  
program **run and  exit no outdata and no error**

when rails/data_phase0.py  change this line 
 ray.init(logging_level=40, local_mode=True, log_to_driver=False)

local_mode=True  from default False to True 

it run ok; 

what happen?

zdx3578, 2021-10-04:
https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md

dotchen, 2021-10-04:
This is odd, last time I saw people having is issue is because of pytorch version.

Can you check what python/pytorch do you have?

zdx3578, 2021-10-05:
'pytorch                   1.4.0           py3.7_cuda10.1.243_cudnn7.6.3_0    pytorch'.   using.  'conda env create -f environment.yaml'

python                    3.7.9                h7579374_0    anaconda

dotchen, 2021-10-05:
Do you see any errors if you set `log_to_driver=True`?

dotchen, 2021-10-05:
Wait, I think I know what your issues are now: you launching 1 CARLA server, but multiple Python workers.
Of course this will fail.

Please use the instructions provided by the README.

zdx3578, 2021-10-06:
yes ,./scripts/launch_carla.sh 1 2000;   data_phase0.py is ok; but data_phase1.py still same  condition 。

dotchen, 2021-10-06:
Any error messages with `log_to_driver=True`?

zdx3578, 2021-10-06:
leaderboard_evaluator.py  debug?

dotchen, 2021-10-06:
How many CARLA servers did you launch?

dotchen, 2021-10-06:
Also, my recommendation to debug:

use one server, one worker, `local_mode=True`.

Set a break point before importlib that loads the agent, and see what happens.

zdx3578, 2021-10-06:
> How many CARLA servers did you launch?

same ,    ./scripts/launch_carla.sh 8 2008  ;

dotchen, 2021-10-08:
Please install the CARLA 0.9.10.1 client as instructed in the `README`. Closing the issue.

closed 2021-10-08 by dotchen