dotchen/WorldOnRails #33: Question  about  data collecting of LBC in this version

YzMark2r, 2021-11-02:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!
Hi, 
I would like to know  How to set stop conditions of LBC  agent during  collecting  data .By reading the code,  I found that the 'q_collecter' and 'random_collecter'   will stop collecting and save the data until the collision. Now, I want to try to train this version of  LBC agent. I think the termination condition for LBC agent to collect data is arriving the destination.But I don't know how to implement it. Can you give me some suggestions or share this part of your code . Thank you , and I will be looking forward to your reply.

dotchen, 2021-11-02:
Hello,

Thank you for your interest in our projects.
The reason we terminate early in case of a collision is that we do not want to collect a bunch of stagnant frames when the ego vehicle is stuck, and of course we will also terminate an episode if the agent arrives at the destination.

All codes needed to collect a dataset is already in this repo. Feel free to ask me if you have any questions.

YzMark2r, 2021-11-02:
I'm glad to receive your reply . I want to collect a dataset to train this version of LBC agent. Can you tell me how to stop colltecting data when the agent arrives the destination in route_xx.xml, or give me some other advice.

dotchen, 2021-11-02:
By default we collect data on randomized routes (see [docs](https://github.com/dotchen/WorldOnRails/blob/release/docs/RAILS.md) for more details).

However if you would like collect data on specific routes, change `evaluate.py` to point agent to the data collecting agent and run it with the route you want to collect data on.

In both cases, the episode terminate if the agent arrives at the destination, by default. Let me know if you having trouble finding relevant codes that handle this.

YzMark2r, 2021-11-02:
Yeah，I have tried to change evaluate.py to point agent to the data collecting agent and run it with the route. Now，I need to set a termination condition to flush data when the agent reaches the destination，like 'if col:' writed in the picture. But I can't. I can only store one frame data by each step in 'def run_step'.[image]

dotchen, 2021-11-02:
Again, you **do not** need to change anything in the code to have it terminate upon arrival at destinations, because this is already set up by default.

The logic of this is not handled in the agent's code, but in the environment. Let me know if this makes sense to you or not.

YzMark2r, 2021-11-02:
Do you mean that scenario will stop automatically when the vehicle reaches the destination？The problem I encountered was a condition for ‘self.flush_data' .I think the code you released is to store the data of each episode after collision. I can't store data with ‘flush_data'  at the end of each episode without replacing this condition with the condition of whether the agent reaches the destination.And I don't know if I'm right.

YzMark2r, 2021-11-02:
And I don't know how to get the parameter of scenario succeed and send the parameter to 'def run_step'.

dotchen, 2021-11-02:
Hi,

Let's look at the problems one by one.

First, do you agree that stopping upon arrival at destinations and stopping upon collisions are not mutually exclusive?

And to answer your question on whether stopping at destination is automatic: yes, as this is the default. At the end of each episode, the agent will be destroyed, and flush all the data to disk in the `destroy` function.

YzMark2r, 2021-11-03:
Thank you for your patient reply. With your suggestion, I solved my problem. Your research is great. I will continue to follow your work.

dotchen, 2021-11-03:
Thank you for liking our work. Feel free to ask me anything if you have trouble running the codes.

closed 2021-11-03 by dotchen