dotchen/WorldOnRails #34: Change bird view to dashboard view

ElliOsm, 2021-11-04:
Hello! 
I want to change the camera I get when I run evaluate_nocrash.py from bird view to a dashboard camera.  
Things I tried to do: 
-I tried to modify the code so I got the view from a dashboard sensor but I couldn't exactly pinpoint what I needed to change and where.
-I thought about recording the session and then viewing it by the sensor I wanted, but the --record argument is only available to the nocrash_evaluator.py and I am having a hard time running it as well.
So can you guide me to the right direction?

dotchen, 2021-11-05:
Hi,

You can change the `visualize` function in the agent file.

ElliOsm, 2021-11-07:
Hello!
Thank you for your quick reply.
What I understood is that, I have to change the `visualize_obs()` function in `image_agent.py`.
Sadly, this is not what I am looking for, what I am looking for is a way to change the sensor that is rendered on screen.

dotchen, 2021-11-07:
Hello,
I am having trouble understanding what exactly you are looking for. Could you post a picture of what you would like to change?

ElliOsm, 2021-11-09:
Hello!
Sorry for the delayed response
Basically what I want  to change is this camera: 
[image]
to this: 
[image]

dotchen, 2021-11-09:
Hello,

In that case you do not need to modify anything. The agents will upload its camera recordings to wandb, so you can simply view the video of frontal facing cameras in your wandb page.

ElliOsm, 2021-11-13:
Ok, thanks a lot for the clarification, I found it at last!

Have a nice day.

closed 2021-11-13 by ElliOsm