dotchen/WorldOnRails #35: Missing label and ego vehicle data in the dataset

eneserciyes, 2021-11-06:
`num_plan=5`, so the last 5 rgb images will not be used for training, but the lbl will because we need them to compute the rewards from future.

_Originally posted by @dotchen in https://github.com/dotchen/WorldOnRails/issues/3#issuecomment-850600844_

eneserciyes, 2021-11-06:
Hello,

Thank you for your amazing work. I just realized that in the current version of the dataset, there are as many label images as there are RGB images which means that the last `num_plan' labels are missing. Is it possible that I am missing something?

dotchen, 2021-11-06:
Hello,

This could be possible, a quick workaround is to further omit the last `num_plan=5` steps. You can also skip the Q-labeling stage and directly use the released pre-computed Q-values.

eneserciyes, 2021-11-06:
Thank you, as I am trying to reproduce your work for the ML Reproducibility Challenge, I went with your first suggestion.

closed 2021-11-06 by eneserciyes