dotchen/WorldOnRails #36: Issues about the map

YzMark2r, 2021-11-19:
Hello, I want to know more about the map.
First, does the channel 'stop' represent the location information of traffic lights ?
Second, What information does the 6 channels of 'waypoints' represent ？
Thank you！
[image]

dotchen, 2021-11-19:
Hello,

> First, does the channel 'stop' represent the location information of traffic lights ?
Traffic lights are dynamically rendered.

> Second, What information does the 6 channels of 'waypoints' represent ？
Each channel represents the desired route corresponding to a high-level command.
(left, right, straight, follow, lane-change left, lane-change right)

YzMark2r, 2021-11-20:
Thank you for your reply. Then, what information does the channels of 'stop' represent ？

dotchen, 2021-11-21:
I was planning to use it for stop/yield signs. End up not doing it for simplicity (running over all stop signs :)