dotchen/WorldOnRails #37: Data collection for LBC

zachren97, 2021-11-22:
Hi,
is there a way to modify the autoagent collector to use the CARLA autopilot as the expert for collecting data instead of using the computational expensive qcollecter agent?

Thank you very much.

dotchen, 2021-11-22:
You can use our released dataset: https://github.com/dotchen/WorldOnRails/blob/release/docs/DATASET.md

zachrenlee, 2021-11-24:
> You can use our released dataset: https://github.com/dotchen/WorldOnRails/blob/release/docs/DATASET.md

Thank you for your prompt reply. I am trying to save the labels as list of bounding boxes instead of images of each label. Thus, I would like to know if it is possible to modify the autoagent collector code to utilize the Carla autopilot as the expert instead of computing the Q values as in the Qcollector agent. I am able enable the autopilot in the map_utils.py by using hero_actor.set_autopilot(True), but I still have not figured out how to get the high-level commands.

dotchen, 2021-11-24:
In that case, the simplest thing you can do is to modify [this agent](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/basic_agent.py)

closed 2022-01-14 by zachren97