dotchen/WorldOnRails #38: Iussues about ego

panpan-77, 2021-12-17:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!

dotchen, 2021-12-18:
Hi,

Thank you for your interest in our project. 

You may customize that in the leaderboard if you have already your customized vehicle imported. here:
https://github.com/dotchen/WorldOnRails/blob/1a81d91d7661822ebb84bf5f6faebd04b2ea7543/leaderboard/leaderboard/scenarios/route_scenario.py#L250-L252

This is for `route_scenario.py` which is for evaluation. Similar changes applies for training scenario

panpan-77, 2021-12-20:
> Hi,
> 
> Thank you for your interest in our project.
> 
> You may customize that in the leaderboard if you have already your customized vehicle imported. here:
> 
> https://github.com/dotchen/WorldOnRails/blob/1a81d91d7661822ebb84bf5f6faebd04b2ea7543/leaderboard/leaderboard/scenarios/route_scenario.py#L250-L252
> 
> This is for which is for evaluation. Similar changes applies for training scenario`route_scenario.py`

panpan-77, 2021-12-20:
> Hi,
>
> Thank you for your interest in our project.
>
> You may customize that in the leaderboard if you have already your customized vehicle imported. here: [...]

hi,thank you for your answer. I have another question.
I already know how to ccontrol the number of pedestrians and cars. But I want to change the scenario. For example, I want to place an obstacle at a point in the map road to see if the ego vehicle under testing can bypass the obstacle or stop.
where I need to find?

dotchen, 2021-12-20:
For this you will need to create your own customized scenario file.

You can read [this](https://carla-scenariorunner.readthedocs.io/en/latest/creating_new_scenario/) for reference.

closed 2022-01-11 by panpan-77