dotchen/WorldOnRails #39: Extending agent for Town03

panpan-77, 2022-01-12:
Hello,

About NoCrash routes.
I am trying to run the agent (pre-trained  RAILS) in the Town03.
I modified part of the code to enable the program to load the Town03. But I found that when I reset the starting point and the ending point (a straight line), the agent can't go straight. Sometimes it always drives to the rightmost Lane in two lanes,even if the start and end points are in the left lane. When the vehicle is driving in the rightmost lane, it cannot overtake if there is a stationary vehicle ahead.
Is there something wrong?
I would appreciate it if you could give me some advice.

Look forward to your reply.

dotchen, 2022-01-12:
Hi,

If you want to run on Town03 you should use the leaderboard pre-trained weights and routes. By design NoCrash is only on Town01 and Town02.

panpan-77, 2022-01-14:
> Hi,
> 
> If you want to run on Town03 you should use the leaderboard pre-trained weights and routes. By design NoCrash is only on Town01 and Town02.

Thank you for your reply. I have two more questions.
1. About use the nocrash pre-trained weights.When driving in town01,the speed of the vehicle is always kept at 5m/s. Could the vehicle speed be changed?If it can, where could I modify it？If it is increased to 16m/s, will it affect the performance of the agent? 
2. If I use the nocrash pre trained weights to run on town 01, will the vehicle overtake under normal circumstances? (I placed an obstacle, but the ego vehicle just stopped.Or a slow NPC vehicle ahead, it just follows the NPC vehicle slowly)

Thank you again for your patience.

panpan-77, 2022-01-14:
Hi,
https://github.com/dotchen/WorldOnRails/blob/d8aa9f7ae67a6b7b71a2fc5ba86bb2a44f221bef/leaderboard/leaderboard/scenarios/scenario_manager.py#L160
I tried to add the following code to this line, but setting the speed in this way makes the agent ineffective and it no longer avoids pedestrians. {self.ego_vehicles[0].set_target_velocity(carla.Vector3D).}

panpan-77, 2022-01-21:
@dotchen Hello,I'm sorry to bother you again, but could you tell me your opinion on the above issue of speed? Thank you very much again

dotchen, 2022-01-21:
Sorry for the late reply. To increase the maximum speed you might also need to tune the granularity of the speed bins since each bin will represent a larger range of velocities (unless by max speed you meant the max speed cap for the test time controller)

closed 2022-03-04 by panpan-77