dotchen/WorldOnRails #41: Training error

ehsannt, 2022-01-20:
First of all, thank you for publishing this work.
I would like to only retrain the Q table again by manipulating the reward function. So, I download the 100GB dataset and ran the data_phase2 module. However, because the dataset that you have provided doesn't have any *.mdb files, lmdb produces an error here because the read-only flag is true and says the no such file or directory. The path that I set is the root of this 1M record dataset.
```python
        # Load dataset
        for full_path in glob.glob(f'{data_dir}/**'):
            txn = lmdb.open(
                full_path,
                max_readers=1, readonly=True,
                lock=False, readahead=False, meminit=False).begin(write=False)
```
Is there anything that I am doing wrong? Thanks.

dotchen, 2022-01-20:
Hi,

Thank you for your interest in our project. Please refer to the [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md#dataset) on this issue.

ehsannt, 2022-01-26:
Thanks for your answer. I'd like to actually use the generated trajectory dataset [100GB] and implement my own data loader but in a similar way to the existing data loader `main_dataset.py`. However, I couldn't understand a few details in this file.

1. What is the number of plans as self.T?
2. Why `spds` and `rots` and `locs` are plural or array in the code below? Aren't they pointing to a single recorded sample in the dataset? Based on what I see in the data.json file and RGB folder, each sample has 12 label images, 6 semantic images for 3 cameras, 6 RGB images for 3 cameras, speed, location, rotation, cmd, action0, action1.

Thank you in advance.

```python
def __getitem__(self, idx):
        
        if not self.multi_cam:
            idx *= len(self.camera_yaws)

        lmdb_txn = self.txn_map[idx]
        index = self.idx_map[idx]
        cam_index = self.yaw_map[idx]
        
        locs = self.__class__.access('loc', lmdb_txn, index, self.T+1, dtype=np.float32)
        rots = self.__class__.access('rot', lmdb_txn, index, self.T, dtype=np.float32)
        spds = self.__class__.access('spd', lmdb_txn, index, self.T, dtype=np.float32).flatten()
        lbls = self.__class__.access('lbl', lmdb_txn, index+1, self.T, dtype=np.uint8).reshape(-1,96,96,12)

        wide_rgb = self.__class__.access('wide_rgb_{}'.format(cam_index),  lmdb_txn, index, 1, dtype=np.uint8).reshape(240,480,3)
        wide_sem = self.__class__.access('wide_sem_{}'.format(cam_index),  lmdb_txn, index, 1, dtype=np.uint8).reshape(240,480)
        narr_rgb = self.__class__.access('narr_rgb_{}'.format(cam_index),  lmdb_txn, index, 1, dtype=np.uint8).reshape(240,384,3)
        cmd = self.__class__.access('cmd', lmdb_txn, index, 1, dtype=np.float32).flatten()

        wide_sem = filter_sem(wide_sem, self.seg_channels)

        # Crop cameras
        wide_rgb = wide_rgb[self.wide_crop_top:,:,::-1]
        wide_sem = wide_sem[self.wide_crop_top:]
        narr_rgb = narr_rgb[:-self.narr_crop_bottom,:,::-1]

        return wide_rgb, wide_sem, narr_rgb, lbls, locs, rots, spds, int(cmd)
```

dotchen, 2022-01-26:
> What is the number of plans as self.T?

This is the number of Bellman updates one does to get the Q/V values.

> Why spds and rots and locs are plural or array in the code below? 

They are array because we asked for array in the code (asked for `self.T`). You can get more details by looking at what the access function does.