dotchen/WorldOnRails #45: compatibility with new versions

klaus99815, 2022-03-09:
Hello, Dian
I am very interested in your project. The new version of Carla enriches many functions and fixes some bugs. The project is great. If the project is compatible with the latest version of Carla, it will have more extensions. Will it be possible to release the training code or model under carla0.9.12+ ?  Or I just wonder if the code will also work in carla0.9.12+ or is it relatively easy to extend it for that?

Thanks, in advance!

dotchen, 2022-03-09:
Hi,

Thank you for your interest in this project. We currently don't have plans to extend it to later CARLA versions, but it should be relatively easy to do so, especially if you do not intent to touch the BEV rendering part of the code.

klaus99815, 2022-04-03:
Hi, @dotchen 
I'm sorry to bother you again.
As you said, the project is relatively easy to be compatible with the latest version of carla0 9.13.  I tried. However, during the evaluation, when running to the second route, the program made an error, as shown in the figure below. In other words, each time the program is started, the vehicle can only run one route and cannot run continuously until the end of the evaluation.
[image]

I'm not very familiar with the code of the leaderboard. I really can't locate the problem. I would appreciate any advice you could give me! Thank you in advance!

dotchen, 2022-04-03:
Can you tell me what agent this is? The collector agent or the image agent?

klaus99815, 2022-04-04:
> Can you tell me what agent this is? The collector agent or the image agent?

the default image agent.

dotchen, 2022-04-04:
Thanks for the info. I would start by first updating the scenario runner and leaderboard to the corresponding latest version if you haven't already done so. Also make sure that the CARLA server does not throw any errors.