dotchen/WorldOnRails #46: about data collecting in data_phase0

Watson52, 2022-04-02:
Hi Chen. Thanks for a lot of open source works and I am trying to follow them.
I jump from the data collecting about LAV as you said there are similarities between them. I found that the random collector return a random control including throttle and steer. I feel confused how could the agent arrived the goal under the random control? Does WOR and LAV use the same agent?

dotchen, 2022-04-02:
Hi,

Thanks a lot for your interest in our projects. 
The distinction between LAV and Rails (WOR) in data collection is: LAV uses an expert;
Rails first use a completely random data to train a vehicle kinematics mode;
it then uses this model to build a crude agent to collect the main traces;
the main traces replay and we use model-based RL offline to distill an agent.

Let me know if you have further questions.

Watson52, 2022-04-02:
Thanks for your reply, it make me clear about the WOR. I think I should relize how the leaderboard and scenario runner work first. I will ask you if I meet problem. Thanks again.

dotchen, 2022-04-02:
Sure thing. Feel free to also email me at dchen@cs.utexas.edu for more questions.