dotchen/WorldOnRails #47: Question about the scenario used in this code

xinshuoweng, 2022-04-04:
Make sure you have read [FAQ](https://github.com/dotchen/WorldOnRails/blob/release/docs/FAQ.md) before posting. 
Thanks!

Hi Dian, 

Thank you for your great work! I have one simple question regarding this file:
https://github.com/dotchen/WorldOnRails/blob/release/assets/all_towns_traffic_scenarios.json

This seems to have much more scenarios than the Carla official scenario files (the old one and also the new one updated 4 months ago):
the old one (with 38894 lines): https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/data/all_towns_traffic_scenarios_public.json
the new one (with 14986 lines): https://github.com/carla-simulator/leaderboard/blob/master/data/all_towns_traffic_scenarios_public.json

So my question is did you create this file (https://github.com/dotchen/WorldOnRails/blob/release/assets/all_towns_traffic_scenarios.json) yourself? Or where did you get this scenario file?

Thanks in advance!
Xinshuo

dotchen, 2022-04-04:
Hi Xinshuo,

Thank you for your interest in our project! 
The scenario file used in this repo is taken from the `leaderboard` branch from the `scenario_runner`, which is the branch that the leaderboard use. 
https://github.com/carla-simulator/scenario_runner/leaderboard/srunner/data/all_towns_traffic_scenarios.json

xinshuoweng, 2022-04-04:
Interesting! I did not notice there are scenario files here as well. Not sure what's the difference from the one in the leaderboard folder though: https://github.com/carla-simulator/leaderboard/blob/master/data/all_towns_traffic_scenarios_public.json

I should probably ask the Carla people about this. But it seems like the one under the leaderboard folder (updated 4 months ago) is newer than the one in the scenario_runner folder (updated 2 years ago), so maybe it is better? Unclear how much performance will be affected using different scenario files to generate data though.

Thanks again!

closed 2022-04-04 by xinshuoweng

dotchen, 2022-04-04:
Sure thing! 

P.S: the one in the master branch might have been updated to accommodate the new CARLA versions (>=0.9.12). So for training models for leaderboard purposes I would recommend sticking to the `leaderboard` branch. I am not 100% sure about this though.